LumaTeam/Luma3DS #1352: Screenshot is not working

Doux91, 2019-12-24:
O3ds
Luma 10.0.1
boot9strap 1.3
Please fix this issue, the screenshot its not working on rosalina menu

Oreo639, 2019-12-24:
I cannot reproduce this. (also did you check `/luma/screenshots`?)
It would be helpful if you could give more info so I can try to reproduce it.

Shadow322, 2019-12-24:
works fine here

[image]

Doux91, 2019-12-27:
> I cannot reproduce this. (also did you check `/luma/screenshots`?)
> It would be helpful if you could give more info so someone can try to reproduce it.
Yes I Opened the folder and its empty, when I use the option screenshot (slow) of rosalina it freezes for seconds and it says Operation succeeded but it does not do anything

Doux91, 2019-12-27:
> works fine here
> 
> [image]

Not for me, I mean rosalina screenshot, not 3ds system screenshot

Shadow322, 2019-12-27:
That is a Rosalina screenshot

TurdPooCharger, 2019-12-27:
You're probably using CTRNAND Luma3DS.
- **[1:] SYSNAND CTRNAND/rw/luma/screenshots**

Doux91, 2019-12-28:
> You're probably using CTRNAND Luma3DS.
> 
> * **[1:] SYSNAND CTRNAND/rw/luma/screenshots**

I think you are right but why is rosalina saving the screenshots in the 3ds?
anyway thanks for help me, I must say the issue is fixed but I would like to have the screenshot on SD or at least change where it saves the screenshot

Shadow322, 2019-12-28:
> I think you are right but why is rosalina saving the screenshots in the 3ds?
> anyway thanks for help me, I must say the issue is fixed but I would like to have the screenshot on SD or at least change [...]

Because the Luma3DS devs refuse to fix it

Doux91, 2019-12-29:
> Because the Luma3DS devs refuse to fix it

I copied all the files from ctrnand to the screenshot of sd card, but this is a big issue

TurdPooCharger, 2019-12-29:
There's nothing wrong with luma. It's your setup. It means your 3DS isn't using Luma3DS found on the SD card.
- **sdmc:/**`boot.firm`

closed 2020-04-15 by TuxSH