LumaTeam/Luma3DS #1354: [Feature request] Region emulation for local multiplayer / download play

elmimmo, 2019-12-26:
Local multiplayer / download play seems to remains region locked. It'd be nice to be able to mix consoles from different regions per game without having to [hard-change the region of the whole console](https://3ds.hacks.guide/region-changing.html), with all it entails.

#240 was about this issue, but was closed until someone learned “more information about what region-free local multiplayer entails”.

PS: Before anyone suggests [Luma Locale Switcher](https://github.com/ericjwg/LumaLocaleSwitcher/releases) note that it does not work at all around the region lock of Local multiplayer / download play.

brunocar, 2020-01-02:
exactly, games like mario kart 7 have download play that makes it unusable between consoles of different regions.

elmimmo, 2020-01-03:
Another example is _The Legend of Zelda: Tri Force Heroes._ It has both local multiplayer (with each console having a copy of the game) and download play (with only one console having the game).

A European console hosting either local multiplayer or download play sessions will remain invisible to a nearby Japanese console, even when the latter has Luma3DS’ “Enable game patching” checked in order to enable out-of-region games.

Similarly, the Japanese console hosting a local multiplayer session of the European game, with its region and language set to the same as the European console’s with Luma Locale Switcher, will remain invisible to the European console, and will outright crash/freeze when attempting to host a download play session, displaying this error message in the bottom screen:

~~~
An error occurred (ErrDisp)

Error type: generic

Process ID: 67115840

Address: 0x00000cea
Error code: 0x000000ce

Press any button to reboot.
~~~

profi200, 2020-01-03:
This is game dependent and there is a chance this will not work at all with some games.

PR #1461 (2020-07-27): Patch DLP client region check
Removes checking whether the system region matches the advertised game region within DLPCLNT:StartScan().
Patches instruction from branching if not equal to doing nothing.
This allows systems from different regions to download play with each other and should resolve #1354.
Tested with a USA N3DS hosting a MarioKart 7 download play session. JP N2DS was able to connect both through the download play application and MarioKart 7.
Feel free to alter style of the changes.
closed 2020-07-27 by TuxSH