LumaTeam/Luma3DS #1356: Got a complete system crashwhen exiting software.

1031982, 2020-01-01:
Not much else to say, other then I included the crash dump here.

[crash_dump_00000001.zip](https://github.com/AuroraWright/Luma3DS/files/4014501/crash_dump_00000001.zip)

Oreo639, 2020-01-01:
What are the steps to reproduce this?
What settings do you have enabled?
Can you follow the template.

1031982, 2020-01-01:
Luma3DS configuration/options:

Screen brightness: (3)

Splash: (before)

Splash duration: (3)

PIN lock: ( )

New 3DS CPU: (off)

--

Autoboot EmuNAND: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

To reproduce :
Open checkpoint, wait for it to populate the installed titles, then exit via home button.

JasonCubed132, 2020-01-01:
Old 3DS
Sys 11.13.0-45E
Luma 10.0.1-plgldr
B9S 1.3
Luma config:
Brightness - 4
Splash - Off
Splash Duration - 1
Pin lock - Off
Check boxes enabled:
- loading external firms
- game patching
- show nand or user string

Can reproduce by suspending checkpoint 3.7.2 using the HOME button then killing via X.

Oreo639, 2020-01-01:
Ah. Sounds like a checkpoint bug and not a luma bug.

1031982, 2020-01-01:
Well, I decided to include it here because a while ago, soon after 10.0.1 was released, I had the system crash the same way on New Super Mario Bros 2.
I just pressed the wrong button so no dump.

However, I will report it with Checkpoint as well.

JasonCubed132, 2020-01-02:
@1031982 Can you open Checkpoint, wait for it to finish loading, and then try to reproduce this? I opened it up on the repo for Checkpoint as being caused by exiting while loading but I want to check if this was the case with you.

1031982, 2020-01-02:
It crashed after it was done loading.
I was already aware of the crash during loading.

MisterPrintf, 2020-02-10:
Most crashes are not actually caused by Luma, but poor coding from other applications. I'd check with the devs of Checkpoint - I've had this crash multiple times but switched back to an earlier version so it hasn't happened since.

xpirad, 2020-04-30:
i have this crash with 3dsident as well.
If i press home and then press x instantly, system will crash.
If i wait a bit and then press the x button, it will be fine.

TuxSH, 2020-05-01:
Not a luma bug

closed 2020-05-01 by TuxSH