LumaTeam/Luma3DS #1357: ErrDisp Type: Generic (Only when Wireless connection is on, the console crashes)

H4un7, 2020-01-03:
Error type:       generic

Process ID:       25
Process name:     nwm
Process title ID: 0x0004013000002d02

Address:          0x00109466
Error code:       0xf9606c02

Ammako, 2020-01-07:
Your console's wireless module is damaged or unseated, or nwm sysmodule is corrupted (most likely the former.)

H4un7, 2020-01-07:
> Your console's wireless module is damaged or unseated, or nwm sysmodule is corrupted (most likely the former.)

So... How can I fix that? And when did it got corrupted anyways? It was working fine a few months ago.

TuxSH, 2020-04-15:
Will have a look into it

TuxSH, 2020-04-15:
Latest commits/oreo's PR should have fixed it.

closed 2020-04-15 by TuxSH

Hrozhek, 2021-03-26:
I have the same problem. After installing module from old 3ds xl to old 3ds problem dissapeared. So change of wifi module may be possible workaround (but start with updating software and after that try to clean contacts before replacement). Fortunately there are a lot of available lots on ali.