LumaTeam/Luma3DS #1359: 3DS CFW black screen, powers on blue light for a few seconds, help!!

chubbysocks8, 2020-01-10:
**System model:**

[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.6.0-39U SysNAND, 11.6.0-39J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. 8.1.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

**Steps to reproduce:**

1.

2.

**Dump file:**

chubbysocks8, 2020-01-10:
When I turn on my console the blue power light goes on for 5 sec and you hear a popping sound.

RocketRobz, 2020-01-10:
Sounds like a hardware issue.
Try checking the ribbons in the console.

chubbysocks8, 2020-01-10:
> Sounds like a hardware issue.
> Try checking the ribbons in the console.

I don't think it is a hardware issue. I tried different boot.firm/boot.3dsx on the root of the sd and once managed to get blue light to stay on but still blank screens.

pilladoll, 2020-01-16:
if you press select+power to power on the console... the luma settings menu appears?

MisterPrintf, 2020-02-10:
The popping sound generally means the top screen is damaged/broken/disconnected. That, or your MCU is fried.
Those on GBATemp would know more so try asking around there.

closed 2020-04-15 by TuxSH