LumaTeam/Luma3DS #1360: [Feature/Request] Rosalina Menu button combination (L + Down + Select) change?

OwOItsDanka, 2020-01-13:
Hey, I'm new here and really just created an account to ask something.
I was in the "finalizing setup" section III of the Homebrew Launcher guide, and I'm having trouble opening the Rosalina menu, because my L button on the 3DS doesn't work. I was wondering if there is anyway to use another button except L? 
Thank you for all your hard work!

Nemos59, 2020-01-18:
Hey, in my knowledge, the only way to change the combo to access the Rosalina menu, is by changing it in the Rosalina menu under "Miscellaneous options...". If your L button does not work, you will not be able to change the combo or access the menu.

OwOItsDanka, 2020-01-18:
Well, that's a shame :/ But still, thanks for answering!

profi200, 2020-01-19:
Replace the config.bin in /luma with this (rename it to .bin first). This has the key combo set to R+Down+Start. Then set the combo to whatever you like.

[config.txt](https://github.com/AuroraWright/Luma3DS/files/4082545/config.txt)

OwOItsDanka, 2020-01-19:
I really thought I'd hit a dead end sooo, damn, thank you so much!

closed 2020-01-19 by OwOItsDanka