LumaTeam/Luma3DS #1368: Ignore ARM11 exception handlers for specific TitleID ?

TheShadowRunner, 2020-02-06:
Hi,
I wish to let the ARM11 exception handlers active but blacklist a certain TitleID (so that the handler ignores faults for that TitleID alone).
Is this possible somehow?
If not, could it be implemented, maybe via a blacklist text file in the /gm9/ folder ?
Thank you.

Oreo639, 2020-02-06:
What is crashing and what is the crash?

TheShadowRunner, 2020-02-07:
Hmm does it matter? I'm just wondering if it can be done. For an app that won't ever be updated and that triggers a benign error. With "ARM11 exception handlers" enables it blackscreens and forces shutdown. With it disabled it works perfectly fine.
So I just wish to blacklist it.
Thanks for considering.

piratesephiroth, 2020-02-07:
Just disable it altogether then. There's no need for it to be enabled all the time.

TheShadowRunner, 2020-02-07:
Hmm I read it's not recommended at all, that it can prevent data corruption when it "legitimately" detects a serious fault (contrary to my scenario), hence why I'd rather keep it enabled + blacklist.

piratesephiroth, 2020-02-07:
In my experience, arm11 exceptions often are minor issues that the 3DS's system doesn't even care about.
The Luma black screen might freeze a game at a very unexpected moment, forcing you tot reboot and making you lose progress as well.

TheShadowRunner, 2020-02-07:
Thanks for these precisions. If such a blacklist isn't possible/ implemented guess it's what I'll do anyway.
I just wish to hear what the developer thinks about it.
Thanks.

TuxSH, 2020-04-15:
https://github.com/AuroraWright/Luma3DS/blob/master/k11_extension/source/fatalExceptionHandlersMain.c#L57 , KCodeSet structure has a titleId member

closed 2020-04-15 by TuxSH