LumaTeam/Luma3DS #1370: Attempting to attach to NWM freezes Rosalina

zhaowenlan1779, 2020-02-12:
**System model:**

New3DS XL

**SysNAND version (+emu/redNAND version if applicable):**

11.13.0-45U

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

v10.0.1.
Also tried 1946941340e5b3fc3afa7943122cbf9c56f9abbb as I cannot build the last commit without updating libctru.

**Luma3DS configuration/options:**

Not related

---

**Explanation of the issue:**

You cannot attach GDB or IDA to the NWM process. Rosalina freezes in this case. (If you are in the menu the entire 3DS freezes; if you are not in the menu, when attempting to enter the menu it freezes)

Attaching to some other processes, like friends, work for me, so I guess the issue is not about my configuration.

**Steps to reproduce:**

1. Enable debugger.
1. Use GDB or IDA to connect to it and try to attach to NWM.
1. Watch as GDB or IDA complains attach failed and the 3DS freeze.

profi200, 2020-02-13:
Of course it freezes, because nwm is needed for the gdb connection.

zhaowenlan1779, 2020-02-13:
Yeah I just realized that. Sorry for being dumb

closed 2020-02-13 by zhaowenlan1779