LumaTeam/Luma3DS #1371: old3DS on old fw - Luma 10.X black screen - Luma 9.1 works

dezem, 2020-02-13:
**System model:**
Old 3DS

**SysNAND version (+emu/redNAND version if applicable):**
4.5.0-10E sysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap

**Luma3DS version:**
10.0 & 10.0.1

**Luma3DS configuration/options:**

Default EmuNAND: ( )
no

Screen brightness: ( )
no

Splash: ( )
no

Splash duration: ( )
no

PIN lock: ( )
no

New 3DS CPU: ( )
no

Autoboot EmuNAND: ( )
no

Use EmuNAND FIRM if booting with R: ( )
no

Enable loading external FIRMs and modules: ( )
no

Enable game patching: ( )
no

Show NAND or user string in System Settings: ( )
yes

Show GBA boot screen in patched AGB_FIRM: ( )
no

Patch ARM9 access: ( )
no

Set developer UNITINFO: ( )
no

Disable ARM11 exception handlers: ( )
no

--

**Explanation of the issue:**
Don't boots, stays in black screen.
If i use Luma3DS version 9.1, it works fine.

**Steps to reproduce:**
The breaking commit is: https://github.com/AuroraWright/Luma3DS/commit/90a43089ba9b9f72ecadca747d7acc0132ccc88f

One commit before, the o3DS boots up normaly: https://github.com/AuroraWright/Luma3DS/commit/1e0baeb30e64255e116b1df87953cd2195e85e05

Another solution:
Update first to firmware 9.X or higher.
So Luma3DS v10.X boots, with no black screen.

TuxSH, 2020-04-15:
That's really weird, I don't see why it would break anything

TuxSH, 2020-04-16:
Should be fixed in latest commit, for now. There wasn't enough space in 4.x to run Rosalina

https://github.com/AuroraWright/Luma3DS/commit/8785fd0236732d7fc20f5595beb881dc47fd265d

closed 2020-04-16 by TuxSH

dezem, 2020-04-17:
Thanks for the fix!
Now it works on older firmware 😄