LumaTeam/Luma3DS #1372: [Feature Request] Backlight Brightness Option

ElDanteDelAnte, 2020-02-14:
It's a bit inconvenient to have to suspend game to adjust screen brightness, plus the high contrast of the Rosalina menu would let me see what I'm doing better as I adjust from indoor to outdoor brightness.

(I'm seeing similar issues listed, but between broken links and closed/outdated threads, I can't tell what the status on this feature is anymore.)

Shadow322, 2020-02-14:
That was killed off via 9.x something as BootNTR has this feature if u downgrade it will show up again via BootNTR

ElDanteDelAnte, 2020-02-14:
[image]
I'm getting conflicting information and I'm confused (I am very new to this, my apologies).
I should clarify that I'm using Rosalina and not BootNTR, and that this is a request for Rosalina. I was linked to this repo when I asked where to suggest Rosalina features.

closed 2020-04-15 by TuxSH