LumaTeam/Luma3DS #1373: Luma crash when opening Super Mario Maker JPN manual

TheShadowRunner, 2020-02-16:
Hey all,
I'd like to report a bug:
When attempting to open Super Mario Maker's Manual, Luma crashes with the following message on the bottom screen:
------
An error occured (ErrDisp)
Error type: generic
Process ID: 67115840
Adress: 0x000005ea
Error code: 0x0000005e
Press any button to reboot.
-----
It's the only game for which opening the manual crashes for me.
I tried to reinstall it multiple times, dump from my own cart 'Super Mario Maker" JPN version, no change, the crash always occurs when attempting to open the manual (and the error on the bottom screen always the same).

Is this a known issue?
Thanks for any help.

Oreo639, 2020-02-17:
Can you try reproducing this on the latest Luma nightly?

TheShadowRunner, 2020-02-17:
> Can you try reproducing this on the latest Luma hourly?

Sure, where may that be downloaded? ^^

TheShadowRunner, 2020-02-17:
Ok, i tested "234-luma3ds-dfd50d9: Merge pull request #9 from Margen67/memes" and the crash still occurs, but this time I have the ErrDisp message as the error description directly.
[image]
Thanks for your help.

TheShadowRunner, 2020-02-18:
Figured it out, corrupted update !
I had no idea updates could affect Manual..
What's strange is only the update to the manual was corrupted, the game patch ver. was displayed on title screen and everything seemed fine.
Anway, mystery solved :)

closed 2020-02-18 by TheShadowRunner