LumaTeam/Luma3DS #1376: Cheats on Pokémon Games are really unstable

DerKlobold, 2020-03-04:
Downloading Cheatcodes from Checkpoint and using them with the Rosalina menu are really unstable. I used only one code at a time and almost always the game just freeues entirely.

I don't know if that's only an issue with the newest luma version or if it was like this before.

Tested it with most Codes from Pokémon Y, Moon, and Alpha Saphire. (Yes I was on the right game version). And from researching I don't think the cheat codes are broken.

ariscop, 2020-03-21:
Having the same trouble, it seems that memory gets remapped with no permissions for some reason, for example 080B5824 from the pokemon y walk through walls code

Using https://github.com/mikewii/Luma3DS/commit/cbc681e1a5e640981a22d5750028b2d268437772 to view mappings shows this, and changing it back to rw- from --- fixes the code

ariscop, 2020-03-21:
Can confirm the same codes work with an ntr plugin, far as i can tell those plugins run on the arm11 where as rosalina is on the arm9

fincs, 2020-03-21:
> rosalina is on the arm9
This is false. Rosalina is an ARM11 custom system module.

Immersion95, 2020-04-21:
Yes I can confirm this behavior. The only solution for now is to use Luma3DS 9.1

TuxSH, 2020-04-29:
what about now?

Theone5000, 2020-04-30:
I can confirm this. Installed the **v10.1.3** version last night and various cheats I used let the Pokemon games freeze. Tested it with **Pokemon X** ID: **0004000000055D00** Version: 1.0/1.5 (It does not matter what version you use)

So what kind of codes freeze the game:
- 30/60 FPS
- No outlines
- Max money
- Use PC anywhere
etc. etc.

Most of these codes just work fine on v9.1. What I do like however in the new version is that both v1.0 and v1.5 codes work together so that's a huge plus. The downside is that you can't use them at all so why bother. Is there a bypass or fix for this?

AverageComet250, 2020-05-19:
I personally find that it varies by game and installation method. on my alpha sapphire digital copy (e-store download) it crashes a lot. on all of the games that I installed via FBI there are less crashes(e.g. omega ruby, ultra moon and all of my vc games). On my Ultra moon copy I get VERY few crashes.

mochizuki-8, 2020-06-23:
I agree with @AverageComet250 that somehow the installation method seems to matter. My cartridge of ultra sun crashes less than FBI installed .cia (personally made from my cartridges) install of Y.

Regardless the Pokemon cheats are incredibly unstable across X/Y USUM ORAS. I would be willing to offer crash logs if any effort would go into fixing this.

CatsLover2006, 2021-02-08:
o3DS or n3DS? I do believe that it's possible the o3DS gets very close to filling its RAM during gameplay.

AverageComet250, 2021-02-08:
I used to play n3ds @CatsLover2006 but that was before I lost my ds because I moved house and put it somewhere.

NE1W01F, 2021-07-15:
i have a n3DS and can say the problem is still there. i tested it on pokemon X and UM

eco9898, 2021-09-02:
I'm running the as of now latest version of checkpoint and luma on my 3ds and the cheats all cause the system to crash. Does anyone know of an alternative way to use cheats on Luma specifically for pokemon X

NewSammyChan, 2021-09-21:
Using a New 3DS and finding crashing on a Pokémon X 

no matter if it’s an all shiny code, instant egg or any other code. Crashes about 10 seconds in

NewSammyChan, 2021-09-21:
Sorry that's not a good enough answer. 

If cheats crash and nothing is going to be done about it. It means for me that CFW is useless. 

Also if it's implemented like this then the feature should be removed.

urherenow, 2021-09-22:
Maybe you should make your own cheats, or talk to the person who made the cheat that you're trying to use.

delilahsrose, 2022-03-20:
I've found that, for some reason, activating the 1.0 version of the "no outlines" cheat, deactivating, then activating the 1.5 version seems to stop crashes entirely. not sure why but it works wonders for me!

delilahsrose, 2022-03-26:
solved it! not sure why it works, but enabling THEN disabling BOTH of the 1.0 & 1.5 no outlines cheat without exiting the luma menu, gives you no outlines without the codes running, so no crashes occur. Hope this helps!

suppai, 2022-06-05:
Closing as a tech support question unrelated to Luma3DS itself; the issue tracker is not for tech support, try Nintendo Homebrew's assistance channels (https://discord.gg/C29hYvh).

closed 2022-06-05 by suppai