LumaTeam/Luma3DS #1380: Can't use Animal Crossing : New Leaf, Welcome Amiibo SpotPass

Gpinchon, 2020-03-22:
**System model:**
New 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**
SysNand, unsure of the version

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap 1.3

**Luma3DS version:**
v10.0.1

**Luma3DS configuration/options:**
Screen brightness: (1)

Splash: (disable)

Splash duration: (1)

PIN lock: (disable)

New 3DS CPU: (L2)

Enable loading external FIRMs and modules: (disabled)

Enable game patching: (enabled)

Show NAND or user string in System Settings: (enabled)

Show GBA boot screen in patched AGB_FIRM: (disabled)

Disable ARM11 exception handlers: (enabled)

**Explanation of the issue:**
Since latest version when using Animal Crossing : New Leaf Welcome Amiibo, I can't access online functionnalities, it askes me to agree with the terms of agreement, which I already agreed to...

**Steps to reproduce:**
1. Launch Animal Crossing : New Leaf, Welcome Amiibo

2. Try and activate SpotPass when clicking "Continue"

ItsPizzaTime1501, 2020-04-15:
Is this console region changed?

TuxSH, 2020-04-25:
Closing for inactivity.

closed 2020-04-25 by TuxSH

Gpinchon, 2020-05-27:
Sorry for the very late response.
No the console region was not changed, it was bough in France and region was not changed since.

AuroraWright, 2020-05-27:
This happens when using a game of a different region. Not a Luma issue, and you can fix it by running this https://github.com/SciresM/EULASetter .