LumaTeam/Luma3DS #1383: Some games crashing on 10.0.x

firelad97, 2020-03-29:
**System model:**
New 2DS

**SysNAND version (+emu/redNAND version if applicable):**
SysNAND 11.13.0-45U

**Entrypoint (How/what you're using to boot Luma3DS):**
Bootstrap9s

**Luma3DS version:**
10.0.1

**Luma3DS configuration/options:**
Default EmuNAND: ( )

Screen brightness: (4)

Splash: (Off)

Splash duration: (1)

PIN lock: (Off)

New 3DS CPU: (Clock)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: (×)

Enable game patching: (×)

Show NAND or user string in System Settings: (×)

Show GBA boot screen in patched AGB_FIRM: (×)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
There seem to be some games that immediately crashes on 10.0.x, no 3DS splash screen even for a moment. Piror to upgrading to 10.0.1, games worked fine on 9.x.x.

Also refer to issue #1336 

Tested games: Persona Q2 & The Legend of Zelda Majora's Mask 3D (will update if I found any more, and both are digital)

**Steps to reproduce:**
1. Launch Persona Q2, Majora's Mask 3D, or a game that may has same issue.

**Dump file:**
 Will post the dump file as soon as I can when I have the access to computer, if not then tomorrow.

TuxSH, 2020-04-17:
Any other games? Can't really repro.

TuxSH, 2020-04-17:
What about this build:

[boot.zip](https://github.com/AuroraWright/Luma3DS/files/4495133/boot.zip)

firelad97, 2020-04-18:
This issue was posted around a half month ago, so it might be fixed by now. I haven't been on my 2DS for a while now, so I'll check.

And dunno why you can't reproduce - I had this issue for over a year and that it never seemed had a fix, for exactly same reason to the issue I referred in my issue. I haven't found any other games that had same problem so far yet.

I kept forgetting to get a log, but it's late right now so I will do that when I have time.

Will try that build later as well.

TuxSH, 2020-04-20:
People say they can't repro with latest commit, closing.

closed 2020-04-20 by TuxSH

firelad97, 2020-05-01:
Just an update. I updated Luma recently and now games I reported previously are working. Still my apologies for not posting the crash dump log when I should had.