LumaTeam/Luma3DS #1384: blue light filter

T0biasCZe, 2020-03-29:
Hello, the bluelight reduce in screenfilters in rosalina menu (R+Ddown+select) work when started, but when the lid close and then open, the filter is off, and you need to turn it manualy again, which is annoying (mainly when you are playing in night and you close it and then open and the blue light shine at your eyes >_<)
steps to reproduce:
press R, down and select to open rosalina
Screen filters
select some level
close rosalina
close hinge
open hinge

Oreo639, 2020-03-29:
https://github.com/AuroraWright/Luma3DS/issues/1268

jbmagination, 2020-03-30:
This is intentional sadly, sleep mode bugs out from it

T0biasCZe, 2020-03-31:
but in the #1268 issue, there is it mainly bugs on 2DS, which isnt closable. what about the new 3ds and new 2ds xl ?

jbmagination, 2020-04-01:
Can you read? The first two lines said it was done on New 3DS
New 2DS XL is probably the same

vertexmachina, 2020-04-08:
If anyone is interested, I built a custom firmware with 7177799021109e9fb582c3a2832c390a679e66cc reverted and it works as expected on my New 2DS XL.

vertexmachina, 2020-04-09:
@jbmagination 
@TuxSH 

For my own purposes, I modified the code to save the filter settings to a file and then read them from disk on boot.

It saves the color temperature (default: 6500) and whether the user wants to have the settings persist across sleeps (default: False).

Then, in Rosalina's main function, it loads the persistence setting from the file and, if it's true, creates the ShellOpen thread. If it's false then it does not create the thread. This way users can opt in to the feature if they want to see if it causes their console to hang on resume. If it does, no worries, reboot and disable persistence. If it doesn't, then they have the feature.

It also loads the saved color temperature from disk so that the setting persists across reboots, although you do have to put the unit to sleep and wake it back up again for it load them. I could not find a good place to automatically apply the saved temperature on boot without causing issues.

I also modified the color filter menu so that you can fine-tune the color temperature in increments/decrements of 500 rather than preset values like they currently are. I have a min of 2000 and a max of 6500 set, so you can do 2000, 2500, 3000, etc, up to 6500.

If this sounds like something you want added then let me know and I will submit a pull request.

jbmagination, 2020-04-10:
Do you mind adding that to my fork, [jbmagination/Luma3DS](https://github.com/jbmagination/Luma3DS)? I’d like to compile it with other stuff I have in my fork

vertexmachina, 2020-04-10:
Sure! https://github.com/jbmagination/Luma3DS/pull/3

TuxSH, 2020-04-15:
Intended, sleep mode wakeup detection caused issues

closed 2020-04-15 by TuxSH