LumaTeam/Luma3DS #1385: [not an actual issue] when will Luma3DS get a new release?

jbmagination, 2020-03-30:
this is not a question to be answered, this is a speculation answer.

well, let's look at recent releases.

it commonly goes v<ver>, to v<ver>.1, occasionally v<ver>.1.1.

normally, it's pretty fast!

so, we can use this common trend to predict about when the next release will come.

it seems to happen sometime within summer. the last 8.0 release in august, the last 9.0 release in july. but 10.0 in june? this may not be a prediction of the next 11.0, but the last 11.0.

so, let's go back to 6.0. the last release of that was near the end of the year, 7.0 in may. i suppose around 7.0 they entirely changed their release schedule. so 7.0 is where we need to start.

7.0 was released in april, 8.0 in june, 9.0 at the end of the year... wait a minute!

this is a repeat, there is a pattern that is being followed, whether it's directly or not.

the first few times are sometime within summer, getting later and later, then it becomes the end of the year normally.

the last 10.0 release is currently in july. 11.0 will be out by august, sometime then, assuming things follow patterns from before.

thoughts on my ridiculous ideas?

Flubbex, 2020-04-01:
Probably never, golden age has ended long ago.

jbmagination, 2020-04-01:
True.

piratesephiroth, 2020-04-01:
Plus there are no new important features to be added, just bugs to be fixed.

profi200, 2020-04-02:
Never say never. When the Switch guys are done they will spend more time on the 3DS again i guess to polish stuff.

And unrelated: I have been working on a big ( ͡° ͜ʖ ͡°) the past week so the 3DS is not that dead homebrewwise.

jbmagination, 2020-04-02:
Alright now my interest is peaked.

TuxSH, 2020-04-15:
When I have fixed a few bugs, I'm fairly sure to know where some of them lie.

Also consider the following: 

- I'm also working on Switch stuff
- I need (long) breaks
- I don't have infinite free time

While I understand that you may be frustrated by the lack of activity, it doesn't excuse you from spamming this issue tracker with garbage.

closed 2020-04-15 by TuxSH