LumaTeam/Luma3DS #1390: Safe Mode causes an ARM9 crash if the combo is entered after the PIN screen or a splash

nicl83, 2020-04-05:
**System model:**

New Nintendo 3DS, KTR-001, E-region

**SysNAND version (+emu/redNAND version if applicable):**

SysNAND 11.13.0-45E

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

Luma3DS 10.0.1

**Luma3DS configuration/options:**

Default EmuNAND: (N/A)

Screen brightness: (4)

Splash: (Off) (also tested with "Before")

Splash duration: (3)

PIN lock: (4 digits) (also tested with no PIN)

New 3DS CPU: (Off)

--

Autoboot EmuNAND: (N/A)

Use EmuNAND FIRM if booting with R: (N/A)

Enable loading external FIRMs and modules: (off)

Enable game patching: (off)

Show NAND or user string in System Settings: (on)

Show GBA boot screen in patched AGB_FIRM: (on)

Patch ARM9 access: (N/A)

Set developer UNITINFO: (off)

Disable ARM11 exception handlers: (off)

--

**Explanation of the issue:**

When I try to boot into Safe Mode by holding L+R+Up+A after the PIN screen or a splash screen, the console crashes with an ARM9 prefetch abort (svcBreak). If I try to boot into Safe Mode by holding L+R+Up+A *before* the PIN screen or a splash (I.E while the console is powered off), the console does not crash, and Safe Mode loads normally. You can find a video of this [here](https://youtu.be/P_VBKZAiQMw).

**Steps to reproduce:**

1. Enable a splash or a PIN code in Luma Config.

2. Turn on the console.

3a. If you have enabled a PIN, enter your PIN, then immediately start holding L+R+Up+A.

3b. If you have enabled a splash, start holding L+R+Up+A when you see the splash.

4. The console will remain on a black screen for a few seconds, before crashing as expected. As buttons are being held down, the console usually shuts off immediately. If you want to get the crash dump, release the buttons when the screen turns black after the PIN or splash screen.

**Dump file:**

[crash_dump_00000000.dmp.zip](https://github.com/AuroraWright/Luma3DS/files/4434094/crash_dump_00000000.dmp.zip)

TuxSH, 2020-04-16:
Can't personally repro it on O3DS but other people tell me it's happening on N3DS.

TuxSH, 2020-04-16:
For future reference, it seems to be an old bug where https://github.com/AuroraWright/Luma3DS/blob/master/arm9/source/large_patches.s#L125 fails with error code 0xC8804465 (mountpoint not mounted)

TuxSH, 2020-04-17:
https://github.com/AuroraWright/Luma3DS/commit/67e28b2a82e782826aabfe79438b7fdd36db7b8c should have fixed it

closed 2020-04-17 by TuxSH

nicl83, 2020-04-17:
Hi Tux, I have been sent a version of boot.firm with the patch applied, and I can confirm that it has fixed this bug. Thanks!