LumaTeam/Luma3DS #1392: ARM11 data abort exception on boss process a few seconds after booting

Latias4Ever, 2020-04-10:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.8.0-41E

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

9.1

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (Off)

Splash duration: (1)

PIN lock: (Off)

New 3DS CPU: (Clock+L2)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

When booting the 3DS, and getting into the Home Menu, after a few seconds, the 3DS crashes with an ARM11 Core 1 exception, with a "data abort" type, "Translation - Section" fault status, and on the "boss" process. Attempting to delete the Home Menu's extdata folder hasn't resolved the issue.

**Steps to reproduce:**

1. Boot 3DS.

2. After 10-30 seconds, crash happens.

**Dump file:**

[crash_dump_00000007.zip](https://github.com/AuroraWright/Luma3DS/files/4461638/crash_dump_00000007.zip)

Flubbex, 2020-04-12:
Had the same problem: https://github.com/AuroraWright/Luma3DS/issues/1382

Turns out my girlfriend had the same problem as well and the above fix apparently doesn't work for her. What does work (for now) was turning on 'Disable ARM11 exceptions' in Luma settings, although now BOSS shows up in the process list as a zombie process. I'm still clueless as to why this is happening but I'm seeing more people talk about it lately. 

Did you by any chance run any homebrew application in the past to disable the 3DS update service? I think that -might- cause it since it messes with the BOSS process. Who knows.

Latias4Ever, 2020-04-12:
No, I haven't run any homebrew applications, or at least I don't remember using them. In fact, my 3DS brought up the update notification every time I went into the home menu when I had it on and working, as it normally should. Until now, it was always in sleep mode with my Ultra Moon cartridge running. To be honest I probably should have just turned it off since I hadn't been using it in ages, but I figured sleep mode would still be fine. Either way, my usage of the 3DS was pretty much the normal usage. I think the most I've done outside the norm is using NTR for RNG manipulation in Ultra Moon/recording on my PC.

I tried to use the "Disable ARM11 Exceptions" setting after seeing that being suggested elsewhere, but the 3DS straight up froze when I tried to return to the Home Menu from my Ultra Moon, so I gave up on that idea.

ItsPizzaTime1501, 2020-04-17:
Disabling arm11 exception handlers won't do shit in most cases, re-enable them and try a safemode update as written [here](https://3dbrew.org/wiki/Recovery_Mode).
If it still doesn't work after updating try a [ctrtransfer](https://3ds.hacks.guide/ctrtransfer.html)

TuxSH, 2020-04-19:
It's just some minor change ninty made where stock ignores it but luma crashes. Ignore it.

closed 2020-04-19 by TuxSH

Yurei29, 2022-06-26:
I have the same problem "an error exception arm11 (core 1) data abort" how did you fix it??