LumaTeam/Luma3DS #1393: Arm11 (Core0) crash when launching .3dsx application

meo136, 2020-04-14:
Whenever I try to launch CtrQuake from homebrew menu, it crashes and says Arm11(Core0) 
Data abort 
And something with translation section 
Im using a old 3ds XL on the newest firmware and luma 10.0.1.
I've just enabled show NAND or user string in System Settings. 
Please. someone. Help me!

meo136, 2020-04-14:
Crash dump coming soon

meo136, 2020-04-14:
[crash_dump_00000024.zip](https://github.com/AuroraWright/Luma3DS/files/4476888/crash_dump_00000024.zip)

TuxSH, 2020-04-16:
Is this only happening with ctrquake? How are you running it?

meo136, 2020-04-16:
No it also happens with CtrHexen II

meo136, 2020-04-16:
What do you mean how am I running it?

TuxSH, 2020-04-16:
Are you using new-hbmenu (https://github.com/fincs/new-hbmenu/releases) ? What about stuff like ftpd (https://github.com/mtheall/ftpd/releases/download/v2.3.1/ftpd.3dsx)?

meo136, 2020-04-17:
Yes I'm using new Hbmenu. What is ftpd?

meo136, 2020-04-17:
Ok. Now I know what ftpd is, but what is it doing?

profi200, 2020-04-17:
He is asking you to try other homebrews like ftpd to narrow down the problem.

meo136, 2020-04-17:
Ok

meo136, 2020-04-17:
Thx for the explanation

meo136, 2020-04-17:
ftpd works fine. The problem only occurs when I try to launch CtrQuake or CtrHexenII

TuxSH, 2020-04-17:
The issue is a CtrQuake/CtrHexenII one then. The author(s) should ensure that these apps are build with the latest toolchain version

closed 2020-04-17 by TuxSH