LumaTeam/Luma3DS #1396: Screen init has been fixed

profi200, 2020-04-25:
I have spend some time reverse engineering GSP code to finally fix this. Corgi3DS was also a huge help. It mostly boils down to waiting for MCU events 24-29 and waiting for the LCDs to indicate they are ready. Not doing so causes random screen init problems like we have been plagued by for years. The fix only works if all homebrew bootloaders respect it because turning screen off for example and booting a FIRM too early can make screen init fail in the booted FIRM.

Because i'm not sure if/how IRQ stuff should be integrated into Luma (which is needed to wait for MCU events) i decided to make an issue instead of a PR. Hopefully that doesn't come across as lazy or something.

Here is the code:
https://github.com/profi200/open_agb_firm/blob/master/source/arm11/hardware/gfx.c#L118

TuxSH, 2020-04-25:
Wow, I certainly didn't expect that, thank you very much!

It's not going into this release but into the next one, for which I have planned to have an IRQ handler for the Arm11 (to prevent MCU i2c polling)

profi200, 2020-04-25:
No worries. It will do for now. When writing that GBA stuff i had screen init problems once more on my N3DS XL so i thought it's time to finally fix this.

hunt-young1, 2023-01-03:
Is it just me or does that link not work anymore

urherenow, 2023-01-03:
Perhaps because this is nearly 3 years old?

profi200, 2023-01-03:
The line you are searching for is now here: https://github.com/profi200/libn3ds/blob/master/source/arm11/drivers/gfx.c#L66

justamii, 2023-01-11:
And what i have to do with this line? i have the same problem with the screen flicking with the splash screen on

NightScript370, 2023-01-30:
> And what i have to do with this line? i have the same problem with the screen flicking with the splash screen on

These lines of code making up a "block of code" is more so pointed here for the developers of the software, who would port over the code from libn3ds/GodMode9 to Luma3DS. It is not beneficial to an end-user.

Kyokuo, 2023-08-29:
So is there any proper workaround for fixing this issue as it perssists a couple years now? Using godmode9 how did u port the fix? Looking at the libn3ds could I maybe compile and run it replacing the splash or is there anything in particular that solves this ? Whatever that may be I really want to use some custom splashes even if it means writing or contributing to the software or some sepparate event handler

profi200, 2023-08-29:
The code base Luma uses right now is missing some things like interrupts that are needed for this so i doubt libn3ds will help here (it's in a pretty messy state so i would assume they don't want to use it anyway). If i recall correctly an attempt was made to port this years ago but it didn't work as expected. GodMode9 has some basic interrupt support at least on ARM11 so it works good enough.

Kyokuo, 2023-08-29:
Does it now? I have been out of the loop for quite some time so I wasn't aware of this. So is it already implemented with godmode9 or you're talking about the possibility of implementing it using it?

profi200, 2023-08-29:
It has been in GodMode9 for years. Not in Luma3DS yet.

panchicleto, 2024-05-28:
How do I get into the code to change the variants?

profi200, 2024-05-29:
You don't. Luma and my codebase are not really compatible and there is some work needed to port that.

I should probably use commit based links from now on. This is the latest one.
https://github.com/profi200/libn3ds/blob/d18c577e10589a52f23b7dafbbfd98c8764529ae/source/arm11/drivers/gfx.c#L303

Meckload, 2024-09-09:
Hi! Just to double check as an end-user with this issue (flickering touch screen). We still wait for this to be implemented in an official Luma update, right?

TuxSH, 2024-10-14:
It'll get fixed when it does. This requires a full rewrite of the boot loader component and I don't want to mess this up

profi200, 2024-10-14:
The way Luma works right now makes this tricky. Specifically the wait for MCU events but i think polling the GPIO would work.

guildmarm, 2024-12-26:
I know this issue is still being worked on. Thankfully on my system it stabilizes and stops flickering after about a minute or two after boot. I'm curious though if there's any temporary solution in the mean time. Also, do we foresee any long term issues for IPS displays facing this issue if splash screens are left enabled and the flicker is left alone?

profi200, 2024-12-26:
Only known workaround is disabling splash screens. If i remember correctly TuxSH attempted to port my code but it didn't really work for him. The codebases are too different. My code requires interrupt support and Luma doesn't have that.

celloh123, 2025-06-27:
having the ghosting + flickering issue even while splash screens are disabled on luma 13.3.2