LumaTeam/Luma3DS #1397: I can't change my rosalina combo after select and start button stopped working

floppydisk89, 2020-04-25:
My select and start button stopped working on my 2ds, is there any way to create a combo to do the start and select function? I can't change the rosalina combo without those buttons.

TuxSH, 2020-04-25:
Change `DEFAULT_MENU_COMBO` in `sysmodules\rosalina\include\menu.h`

Here's a build where I changed the default menu combo to just "R" (GH file upload seems to be irresponsive): https://nopaste.me/#bZ6W_1Ty5O_OxvFQP0Z5OA

floppydisk89, 2020-04-25:
I'm unable to compile, would you compile one with the combo Left trigger + Dpad Right + A?
Is is possible to implement combos to press start and select with other combos eg: Dpad Right + b for start and Dpad right + b for select?

floppydisk89, 2020-04-25:
I did manage with your custom boot.firm to get to rosalina, thank you! I was in a rush and didn't think right when i was reading, and it was a shame for me because i am a programmer. Sorry for the comment above...

Do you think rosalina could have custom combos to achieve SELECT and START button actions?

TuxSH, 2020-04-25:
> The custom boot.firm doesn't also save the new combo I added

There's an option in the Miscellaneous submenu, "Save settings" that you need to use. Once you use it, it should retain the new combo.

> it makes my console to take ages to shutdown.

working on fixing it

TuxSH, 2020-04-26:
Apparently this is a thing too: https://3ds.eiphax.tech/config.html

closed 2020-04-26 by TuxSH