LumaTeam/Luma3DS #1398: Memdump feature crashes on app with large code and heap sections

zoogie, 2020-04-25:
**System model:**
New 3DS
[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**
11.13.0-45U Sysnand
[e.g. 11.13.0-45U SysNAND, 11.13.0-45E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**
boot9strap
[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**
10.1 (10.0.1 also has the issue)
[e.g. v10.0.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/0543c208fd154e6326ea5da8cbf66ffcbdef010c]

**Luma3DS configuration/options:**
everything is default settings
Default EmuNAND: ( )

Screen brightness: (4 )

Splash: (off )

Splash duration: (1 )

PIN lock: (off )

New 3DS CPU: (off )

--

Autoboot EmuNAND: (not applicable )

Use EmuNAND FIRM if booting with R: ( not applicable )

Everything is unchecked
Enable loading external FIRMs and modules: (  )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch Arm9 access: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

--

**Explanation of the issue:**
When trying to dump code or heap memory with Smilebasic (US appvers. 3.6.0), the system will crash (fs module). Note that this game has relatively large code and heap sections (29MBs and 82MBs respectively). I can work around this issue (sorta) by either limiting the dump size in the source code to < 16MBs or chunking filewrites to 1MB at a time. For some reason, the code dump will still crash at the last MB with the 1MB chunking workaround.

Additional note: Version 3.6.0 of smilebasic added a screenshot dump feature that's triggered by the Select button (which is also the luma memdump button of course). Repeating the memdump operation on appvers. 3.3.2 produced exactly the same issue as 3.6.0, so we can rule out the screenshot filedump possibly interfering with the luma dump.

**Steps to reproduce:**

1. I've noticed the same behavior on multiple 3ds firmwares and Smilebasic app/region versions. Old3ds as well.

2.

**Dump file:**

[crash_dump_00000016.zip](https://github.com/AuroraWright/Luma3DS/files/4534288/crash_dump_00000016.zip)

TuxSH, 2020-04-25:
Does latest commit also has the issue?

That being said, I would recommend using gdb instead

zoogie, 2020-04-26:
Same result on latest commit.

TuxSH, 2020-04-26:
Your crash dump indicates that the kernel panics because the buffer that it tries to dump doesn't have correct permissions/is not mapped. This could be a rare race condition.

Is it crashing with other games?

zoogie, 2020-04-27:
None that I can remember off the top of my head. Then again, I've never seen a game even come close to the gargantuan code and heap sizes of SmileBasic.

When I chunk the filewrites of 1MB (or anything smallish) the entire 82MB heap successfully dumps. The code section gets to the last megabyte then gives that same crash screen. I don't know if that info helps any.