LumaTeam/Luma3DS #1399: It takes time to turn off the console

DragonNetAlfa, 2020-04-26:
**System model:**

[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.13.0-45U SysNAND, 11.13.0-45E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. v10.0.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/0543c208fd154e6326ea5da8cbf66ffcbdef010c]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (X)

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch Arm9 access: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: (X)

--

**Explanation of the issue:**

**Steps to reproduce:**

1. Is it normal that in version 10.1 it takes to turn off the console? Update 2 consoles and both take time to shutdown.

2.

**Dump file:**

RocketRobz, 2020-04-26:
How long did it take?

DragonNetAlfa, 2020-04-26:
16 sec.

DragonNetAlfa, 2020-04-26:
> ¿Cuánto tiempo tomó?

16 sec

michelrt1992, 2020-04-26:
yeah, i have the same issue! 10 to 16 seconds to turn off :(

RocketRobz, 2020-04-26:
This also happened on the previous version for me, but sometimes.
Maybe it's normal?

michelrt1992, 2020-04-26:
> This also happened on the previous version for me, but sometimes.
> Maybe it's normal?

i returned from 10.1 to 10.0.1 and it turns off immediately, is weird.

magyk188, 2020-04-26:
Before 10.1 update it was around the same time as stock 3ds shutdown, after update it takes around 15 secs. Not too big of an issue but would like for it to shut down faster.

profi200, 2020-04-26:
10.0.1 also had unusual long poweroff times but it was way less than 16 seconds.

TuxSH, 2020-04-26:
I'm currently investiga
ting the matter.

TuxSH, 2020-04-26:
Should be fixed now; observing shutdown times of less than 3 seconds

This was a combination of 3 issues:

* An issue dating back from around 2016: pxi sysmodule not actually exiting, making pm wait more. https://github.com/AuroraWright/Luma3DS/commit/eb37ac4142d21010f1bbc8807336b1588adc703c . Similar thing for Rosalina as well
* A regression from the WiFi forcing PR: unnecessarily calling acInit before main, and keeping it forever because of the issue above. I believe this caused boot delays as well. https://github.com/AuroraWright/Luma3DS/commit/1d8b793cf760887c9519dd0ebdff5c95b928d2aa
* A 10.0.x regression: waiting for a thread that never exited. https://github.com/AuroraWright/Luma3DS/commit/9d6299579912a321073343ed25f6abee850f87ee

closed 2020-04-26 by TuxSH

DragonNetAlfa, 2020-04-26:
> Debería arreglarse ahora; observando tiempos de apagado de menos de 3 segundos
> 
> Esta fue una combinación de 3 problemas:
> 
> * Un problema que se remonta a alrededor de 2016: pxi sysmodule no sale realmente, haciendo que pm espere más. Algo similar para Rosalina también. [eb37ac4](https://github.com/AuroraWright/Luma3DS/commit/eb37ac4142d21010f1bbc8807336b1588adc703c)
> * Una regresión del WiFi forzando PR: llamando innecesariamente a acInit antes de main, y manteniéndolo para siempre debido al problema anterior. Creo que esto también causó retrasos en el arranque. [1d8b793](https://github.com/AuroraWright/Luma3DS/commit/1d8b793cf760887c9519dd0ebdff5c95b928d2aa)
> * Una regresión 10.0.x: esperando un hilo que nunca salió. [9d62995](https://github.com/AuroraWright/Luma3DS/commit/9d6299579912a321073343ed25f6abee850f87ee)

Already update luma again and the same error persists. Takes 16 sec to shutdown

TuxSH, 2020-04-26:
I haven't made a new release nor did I update the current one. Here is a build of the lastest commit: 
[boot.zip](https://github.com/AuroraWright/Luma3DS/files/4536463/boot.zip)

DragonNetAlfa, 2020-04-26:
> No he hecho una nueva versión ni actualicé la actual. Aquí hay una compilación de la última confirmación:
> [boot.zip](https://github.com/AuroraWright/Luma3DS/files/4536463/boot.zip)

Update through luma update and the new version that appeared to me was this, I do not know if it is the same as the .zip
v10.1 -31891ef

DragonNetAlfa, 2020-04-26:
> I haven't made a new release nor did I update the current one. Here is a build of the lastest commit:
> [boot.zip](https://github.com/AuroraWright/Luma3DS/files/4536463/boot.zip)

I just tried the version you provided me and indeed the problem was solved. Will you update it on the luma page? And thanks for your responses and support.

Oreo639, 2020-04-26:
These changes are not currently in the latest release. Tux needs people to test the changes to make sure that it issue is properly solved.