LumaTeam/Luma3DS #1404: Cheats don't work in 10.1.1

Someguy1519, 2020-04-27:
-- Please fill in the placeholders.-->
**System model:**

New 3DS XL

**SysNAND version**

11.13.0-45U

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

v10.1.1

**Luma3DS configuration/options:**

Screen brightness: (1)

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Enable loading external FIRMs and modules: (x)
Enable game patching: (x)
Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: (x)

Patch Arm9 access: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

--

**Explanation of the issue:**

Cheats don't work when I select them in the Rosalina Menu. The selected cheats turn red and nothing happens in-game. They worked fine in 10.1

**Steps to reproduce:**

1. Go into a game with cheats installed

2. Press L+Down+Select to Access the Rosalina menu

3. Go to Cheats and select the cheats

TuxSH, 2020-04-27:
Which ones?

Someguy1519, 2020-04-27:
> Which cheats are turning red?

In Pokémon Super Mystery Dungeon USA version the cheats for 999 Goldbars, Full Belly, Infinite PP, Item quantities x99, Max EXP for Slots 1 through 3, Max deposited money all turn red.

TuxSH, 2020-04-27:
Can you copy paste the actual lines that are turning red, please?

Someguy1519, 2020-04-27:
> Can you copy paste the actual lines that are turning red, please?

What do you mean by lines?

profi200, 2020-04-27:
The cheat codes which don't work from the cheats.txt file.

Someguy1519, 2020-04-27:
> The cheat codes which don't work from the cheats.txt file.

There is no cheats.txt file. I can't seem to find it anywhere on my SD card

profi200, 2020-04-27:
It has to be there or you would not see anything.

https://github.com/AuroraWright/Luma3DS/wiki/Cheat-Engine

Someguy1519, 2020-04-27:
> It has to be there or you would not see anything.
> 
> https://github.com/AuroraWright/Luma3DS/wiki/Cheat-Engine

I checked Luma/titles/titleidofthegame/ and the cheats.txt file is not there for some odd reason.

Someguy1519, 2020-04-27:
The cheats do show up in the Rosalina Menu despite the cheats.txt file being missing

profi200, 2020-04-27:
You probably have installed Luma to NAND. Boot into GodMode9 and open drive "[1:] SYSNAND CTRNAND". The luma folder should be in there. You can then copy cheats.txt to the SD card if you found it.

Someguy1519, 2020-04-27:
> You probably have installed Luma to NAND. Boot into GodMode9 and open drive "[1:] SYSNAND CTRNAND". The luma folder should be in there. You can then copy cheats.txt to the SD card if you found it.

I checked there and it's not there either

Someguy1519, 2020-04-27:
I've tested cheats on another game and I get the same result: Selected cheats highlighted in red after getting the cheats from Checkpoint. I think cheats in general are broken in 10.1.1 not just for specific games.

TuxSH, 2020-04-27:
I am asking once more: what are the contents of the lines that are being highlighted in red?

by line I mean = two sequences of 8 letter or digits, eg. DC000000 1234ABCD

Someguy1519, 2020-04-27:
> I am asking once more: what are the contents of the lines that are being highlighted in red?
> 
> by line I mean = two sequences of 8 letter or digits, eg. DC000000 1234ABCD

How do I check that without a cheats.txt file?

commit 2b23be8f (2020-04-27): rosalina/cheats: fix v10.1.1 regression where cheats were not working at all. Fixes #1404
closed 2020-04-27 by TuxSH