LumaTeam/Luma3DS #1405: Luma 10.1.1 reports as 10.1

bmn001, 2020-04-27:
**System model:**
New 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**
Sys 11.13.0-45U

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap 

**Luma3DS version:**

10.1.1

**Luma3DS configuration/options:**

Pretty much all Defaults

--

I just updated to Luma3DS 10.1.1.  Powered down, rebooted into system menu, opened up the Rosalina menu, and down at the bottom it says "Luma3DS v10.1"

I know it's a minor release, but I suspect the intended behavior is for this screen to say "Luma3DS v10.1.1"

**Steps to reproduce:**

1. replace boot.firm and boot.3dsx on the root of the SD.  Copy boot.firm to ctrnand.
2. full power down
3. power up
4. Wait for system menu to appear, fire up Rosalina menu, see old version number.

**Dump file:**
n/a

TuxSH, 2020-04-28:
oops

TuxSH, 2020-04-28:
Will release 10.1.2 anyway

closed 2020-04-29 by TuxSH