LumaTeam/Luma3DS #1407: Regression: Input Redirection sleep lock #904

isavegas, 2020-04-28:
**System model:**
New 3DS KTR-001 

**NAND versions:**
* 11.13.0-45U
* No EmuNAND

**Entrypoint:**
Boot9Strap

**Luma3DS version:**
v10.1.2-44cd392 stable via Luma Updater

**Luma3DS configuration/options:**

Default EmuNAND: ( )
Screen brightness: ( 1 )
Splash: ( off )
Splash duration: ( 1 sec )
PIN lock: ( off )
New 3DS CPU: ( clock + L2 )

Autoboot EmuNAND: ( )
Use EmuNAND FIRM if booting with R: ( )
Enable loading external FIRMs and modules: ( )
Enable game patching: ( )
Show NAND or user string in System Settings: ( x )
Show GBA boot screen in patched AGB_FIRM: ( )
Patch Arm9 access: ( )
Set developer UNITINFO: ( )
Disable Arm11 exception handlers: ( )

**Explanation of the issue:**
Regression of bug causing #904. Must disable input redirection before putting device to sleep or it will refuse to wake up without a hard reboot. Perhaps input redirection should be automatically disabled when the device is going to sleep, at least until the core issue is resolved?

**Steps to reproduce:**
1. Enable Input Redirection via Rosalina.
2. Put device to sleep by closing the 3DS.
3. Attempt to wake device by opening the 3DS.
4. The device is now locked in sleep mode.

**Dump file:**
N/A

TuxSH, 2020-04-29:
I closed the old issue because it was very old.

> Perhaps input redirection should be automatically disabled when the device is going to sleep

I tried doing exactly just that a few days ago and it didn't work. I don't want to mess with sleep mode so I'll leavei it as that.

closed 2020-04-29 by TuxSH

kanjieater, 2020-05-17:
Is there are recommended workaround if this is a "wontfix"?

urherenow, 2020-05-17:
The workaround is in the issue. Disable input redirects, if you’re going to put it in skeep mode.