LumaTeam/Luma3DS #1408: Unknown Arm11 exception when loading ds mode/safe mode, exit from settings or shutdown 3ds

S4n1c, 2020-04-28:
**System model:**
New 3DS
[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**
11.13.0-45E
[e.g. 11.13.0-45U SysNAND, 11.13.0-45E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**
Fastboot3DS
[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**
v10.1.2
[e.g. v10.1.2 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/0543c208fd154e6326ea5da8cbf66ffcbdef010c]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( 4 )

Splash: ( Before payloads )

Splash duration: ( 3 seconds )

PIN lock: ( No )

New 3DS CPU: ( Clock + L2 )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( X )

Show NAND or user string in System Settings: ( X )

Show GBA boot screen in patched AGB_FIRM: ( X )

Patch Arm9 access: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

--

**Explanation of the issue:**
After updating to v10.1.1, when I load ds mode or safe mode, exit from system settings or shutdown the 3DS i get a "An exception has occured" screen saying this: 
Processor: Arm11 (core 1)
Exception type: data abort
Fault status Translation - Section
Current process: pxi (0000000000000000)
And some some hex values(?)

**Steps to reproduce:**

1. Update to v10.1.1/v10.1.2

2. ???

**Dump file:**

[crash_dump_00000001.zip](https://github.com/AuroraWright/Luma3DS/files/4548600/crash_dump_00000001.zip)

TuxSH, 2020-04-29:
How did you update? Using https://github.com/AuroraWright/Luma3DS/releases/download/v10.1.2/Luma3DSv10.1.2.zip ?

S4n1c, 2020-04-29:
Yes, i did update manually, using the latest release zip from github. I also tried updating with universal updater, with the same result.

TuxSH, 2020-04-29:
I'm having trouble reproducing the issue on my end, on the latest sysver on my N2DS... even if I see what kind of issue it is

S4n1c, 2020-04-29:
I don't know how it happend, i just updated it and it started doing this. This happens only on 10.1.1+ and not on 10.1 so it should be an issue with the changes made in 10.1.1. Also i can try and find the hourlies for the changes beetween 10.1 and 10.1.1, if that helps.

S4n1c, 2020-04-29:
It could also be a Fastboot3ds problem, now that i think about it.

TuxSH, 2020-04-29:
Doubt it

TuxSH, 2020-04-29:
@S4n1c 
[boot.zip](https://github.com/AuroraWright/Luma3DS/files/4552818/boot.zip) does this fix it for you?

S4n1c, 2020-04-29:
It 100% does! Thanks for the help!

closed 2020-04-29 by S4n1c