LumaTeam/Luma3DS #1409: Crash on Shutdown on 10.1.2

Rowan990, 2020-04-29:
**System model:**

New 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**

Sys 11.13.0-45E

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

10.1.2-44cd392

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (4)

Splash: (Off )

Splash duration: (1)

PIN lock: (Off)

New 3DS CPU: (Off)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: (Off)

Enable game patching: (On)

Show NAND or user string in System Settings: (On)

Show GBA boot screen in patched AGB_FIRM: (Off)

Patch Arm9 access: (N/A)

Set developer UNITINFO: (Off)

Disable Arm11 exception handlers: (Off)

--

**Explanation of the issue:**

After updating to the latest stable version (10.1.2), I'm experiencing frequent crashes on shutdown. Attached a few dump files.

[Dumps.zip](https://github.com/AuroraWright/Luma3DS/files/4553148/Dumps.zip)

**Steps to reproduce:**

1. Shut down the system.

2.

**Dump file:**

TuxSH, 2020-04-29:
Does this build: https://github.com/AuroraWright/Luma3DS/files/4552818/boot.zip fix it?

TuxSH, 2020-04-29:
(duplicate of https://github.com/AuroraWright/Luma3DS/issues/1408)

Rowan990, 2020-04-29:
Yes, it does indeed fix the issue. Thanks a lot! And apologies for having opened a duplicate issue.

closed 2020-04-29 by Rowan990

TuxSH, 2020-04-29:
Np; Actually I needed more than two affected people to test :)