LumaTeam/Luma3DS #1410: [Crash] ARM11 data abort exception when closing Senran Kagura 2: Deep Crimson from Home Menu.

Cueropedo, 2020-05-02:
System model:
New 3DS XL

SysNAND version (+emu/redNAND version if applicable):
11.13.0-45U

Entrypoint (How/what you're using to boot Luma3DS):
Followed 3ds.hacks.guide, so I guess it's Boot9Strap.

Luma3DS version:
v10.1.3 (This did happen as well with 10.0.1)

Luma3DS configuration/options:

Screen brightness: (2)

Splash: (Off)

Splash duration: (1)

PIN lock: (Off)

New 3DS CPU: (Off)

--

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

--

Explanation of the issue:
Luma3DS crashes after closing the game on the Nintendo 3DS, it doesn't happen with all missions, but there are some that does this.
One of them it's the "At the Height of Battle" mission of Chapter 2.

Steps to reproduce:

1. Play the game normally until you finish one "bugged" mission such as "At the Height of Battle" mission of Chapter 2. . 

2. After finishing the mission, go to the Ninja Room and then go to Home Menu to close the game. After doing that, you will get the crashing screen.

Dump file:
[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/4568859/crash_dump_00000000.zip)

TuxSH, 2020-05-02:
There's a NULL deref in the game when you're closing it (data abort with FAR=0), most likely the crash happens silently on stock.

Not a luma bug

closed 2020-05-02 by TuxSH

Cueropedo, 2020-05-03:
Thank you, sorry for the inconveniencie, if I disable the ARM11 exception handler it should stop giving that crash?

TuxSH, 2020-05-04:
it should, but I don't recommend using that option otherwise.