LumaTeam/Luma3DS #1412: N3DSXL Will not boot on latest stable

hunttung, 2020-05-04:
I Replaced my old version of luma with Luma3DS 10.1.3 and my screens will not come on, it just hangs on a blue light. I am on firmware version 11.13 and when i went back to an old hourly it worked fine again. i heard about the IPS thing and how it was fixed but i guess it was not since my 3DS will not boot at all on 10.1.3. I tried taking the SD Card out and putting it back and powering off and back on but neither helped at all. Please look into this. i can provide screenshots or videos if needed.

TuxSH, 2020-05-05:
Are you using this release archive? https://github.com/AuroraWright/Luma3DS/releases/download/v10.1.3/Luma3DSv10.1.3.zip ?

Ammako, 2020-05-05:
Also, just to rule that out as a possibility, make sure the file isn't getting corrupted at some point before or after making it to your SD card, by comparing the checksum of the file on your SD card with the one from the archive. The file being corrupted can cause the system to be stuck on blue light+black screen on boot.

hunttung, 2020-05-05:
> Are you using this release archive? https://github.com/AuroraWright/Luma3DS/releases/download/v10.1.3/Luma3DSv10.1.3.zip ?

> Also, just to rule that out as a possibility, make sure the file isn't getting corrupted at some point before or after making it to your SD card, by comparing the checksum of the file on your SD [...]

i did use that release archive and i did check with a checksum viewer and there was no change between the computer and when in my sd card

closed 2020-05-05 by TuxSH