LumaTeam/Luma3DS #1413: An exception occured: PXI when rebooting / shutting down (9942e8b)

grimson73, 2020-05-05:
**System model:**

I do have 2 x New 3DS XL which I keep identical

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.13.0-45U SysNAND, 11.13.0-45E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap

**Luma3DS version:**
https://github.com/AuroraWright/Luma3DS/commit/9942e8b

--

**Explanation of the issue:**
When shutting down / Reboot:

An Exception occured
Exception type: data abort
Fault status: Translation - Section
Current Process: pxi (0000000000000000)

**Steps to reproduce:**

1. https://github.com/hax0kartik/luma-hourlies/releases/tag/46-luma3ds-9942e8b
Place boot.firm in root SD card

2. Reboot N3DSXL to load hourly firmware and then to reproduce:
 - Shut down console Rosalina menu / Power button
 - Reboot console rosalina menu

**Dump file:**

[N3DSXL1.zip](https://github.com/AuroraWright/Luma3DS/files/4580342/N3DSXL1.zip)
[N3DSXL2.zip](https://github.com/AuroraWright/Luma3DS/files/4580343/N3DSXL2.zip)

TuxSH, 2020-05-05:
[boot.zip](https://github.com/AuroraWright/Luma3DS/files/4581985/boot.zip)
Kartik's build stuff is having hard-to-solve issues, build it yourself. Here's my own build of the latest commit.

closed 2020-05-05 by TuxSH

grimson73, 2020-05-05:
Thank you! your provided build does work as intended :).