LumaTeam/Luma3DS #1416: [Feature Request] More options for searching values in RAM viewer

Konsonanz, 2020-05-06:
Hello,
first off, thanks a lot for this great project and the continuous improvements! I'm using and enjoying it on my system quite a lot.

I'm using the RAM viewer right now for editing values on the heap, and I would like to request the following improvements:
- allow searching for values greater in size then only one byte
- ability to search for the next match forward, and optionally backwards

The background here is that I'm editing an inventory, where every entry/item is 8 bytes and "stacked" one after another, with the first 4 bytes being the item-id. Instead of manually scrolling through searching for the desired item-id, it would be preferable to just jump to it. Right now I can only search for a single byte, which does not work reliably (the inventory is a pretty big heap-region with ~300 entrys).

Thank you for your consideration!

TuxSH, 2020-05-08:
Thanks :)

I'll maybe add that later, but the memory viewer code is kind of a mess. Plus, the reason I've never touched the memory viewing code is that the GDB stub exists (I've also implemented a fast impl of memory search on its side): https://sourceware.org/gdb/onlinedocs/gdb/Searching-Memory.html

I believe using GDB suits your needs more.

Konsonanz, 2020-05-09:
Thanks for the reply!
I agree, this improvement would be purely for quality of live (being able to edit memory on the device itself is already pure awesomeness; the improved search / searching without first attaching a second device would have just been the "icing on the cake").

I've played around with the GDB stub the last day and it works perfectly as well for me. I'd be fine with closing this isse then.

closed 2020-05-14 by TuxSH