LumaTeam/Luma3DS #1418: Unable to open PKSM file while Luma3DS is running on Citra for Android

aecarvalho, 2020-05-07:
It seems like I can open Luma3DS launcher on Citra for Android but if I try to open PKSM from within Luma3DS then the system just freezes until I exit the app, any solution would be much appreciated. 
[image]

Shadow322, 2020-05-07:
This does not belong here

Oreo639, 2020-05-07:
The homebrew launcher doesn't have an interface for launching 3dsx file with Citra. It is designed for running 3dsx Homebrew on hardware. This isn't a bug.
You can load the 3dsx for PKSM directly in Citra.

Keep in mind as Shadow said this is for reporting issues with Luma3DS, not issues in Citra.

aecarvalho, 2020-05-07:
Actually it isn't a Citra issue, loading PKSM directly into Citra gives me the following error;
[image]

The above report is a freeze happening when PKSM is opened within Luma3DS

Oreo639, 2020-05-07:
The Homebrew Launcher and Luma3DS are not the same thing. You cannot run Luma3DS in Citra, it is impossible.
The Homebew Launcher can be run, but it doesn't support Citra and you can't launch apps from it.
PKSM performs a check for the Rosilina sysmodule, that is not a bug in luma, PKSM only supports running on real hardware with Luma3DS.

aecarvalho, 2020-05-07:
> Keep in mind as Shadow said this is for reporting issues with Luma3DS. Yet what you are showing is The Homebrew Launcher running in Citra.

Could you show me an instance of you running PKSM from Luma3DS just for reference?

aecarvalho, 2020-05-07:
> The Homebrew Launcher and Luma3DS are not the same thing. You cannot run Luma3DS in Citra, it is impossible.
> The Homebew Launcher can be run, but it doesn't support Citra and you can't launch apps from it.
> PKSM performs a check for the Rosilina sysmodule, that is not a bug in luma, nor is it a bug in PKSM, nor is it a bug in CItra. PKSM only supports running on real hardware with Luma3DS.

Thanks, much appreciated

closed 2020-05-07 by aecarvalho