LumaTeam/Luma3DS #1420: [Feature Request] Delete, disable or hide "Healt & Safety Information"

wuniversales, 2020-05-09:
This is something that no one really uses.
It would be nice to be able to hide it, or directly remove it from the nintendo 3ds.

profi200, 2020-05-09:
No, it's just on the HOME menu. Why does it bother you? If you can't live with it just uninstall it with FBI.

Note:
Not recommended to uninstall system titles and they will come back with updates.

Nuck-TH, 2020-05-09:
Shove it into folder with other system apps. This is not function of **CFW**.

closed 2020-05-09 by AuroraWright

Ammako, 2020-05-09:
If you care enough...

- Dump the CXI (you can do this via GM9) and [extract contents of ExeFS and RomFS](https://github.com/ihaveamac/3DS-rom-tools/wiki/Extract-a-game-or-application-in-.cxi%2C-.cfa-or-.ncch-format)

- [Edit icon in exefs to set visibility flag to 0](https://www.3dbrew.org/wiki/SMDH#Flags)

- [Rebuild the CIA](https://github.com/ihaveamac/3DS-rom-tools/wiki/Rebuilding-a-game-or-application-in-.cia-format). Set a version number equal or higher than the latest version of the app, idk offhand but FBI will tell you.

Install the edited app, purge home menu icon cache, you're good to go.

Shadow322, 2020-05-09:
> This is something that no one really uses.
> It would be nice to be able to hide it, or directly remove it from the nintendo 3ds.

I use it all the time