LumaTeam/Luma3DS #1424: The machine is totally broken after opening the manu

JudyLucy, 2020-05-18:
I have a new 3dsll and my system is 11.13 and was updated online. My luma is 10.1.3 now. Before updating the system, I can open the luma menu by pressing L+down+select. At that time, my system was 11.6 and my luma was 8.1 and was also downloaded here. My machine can work well now without opening the menu, but after pressing the L+down+select, it totally broken. I want to control my machine with the help of another Xbox joy-con, so it's necessary for me to open it every time I play the games. Is there a way of fixing it? Thanks for reading my bug report.

TuxSH, 2020-05-21:
I think your buttons are stuck and may need repair.

In the meanwhile, can you try this build I haven't released yet? 
[boot.zip](https://github.com/LumaTeam/Luma3DS/files/4665194/boot.zip)

JudyLucy, 2020-05-22:
Oh, you've reminded me of my broken camare, tough I don't know whether it has anything with that frozen manu. I will send back my feedback as soon as possible. Thanks for replying my letter.

JudyLucy, 2020-05-22:
Problem settled! I love you my dear sister. Since it is still confidential, I will just keep it for personal use.

Shadow322, 2020-05-22:
> Problem settled! I love you my dear sister. Since it is still confidential, I will just keep it for personal use.

What is confidential?

JudyLucy, 2020-05-22:
> > Problem settled! I love you my dear sister. Since it is still confidential, I will just keep it for personal use.
> 
> What is confidential?

The file of course. The file above mentioned hasn't been released officially, so it is confidential and I will keep it just for personal use.

Shadow322, 2020-05-22:
> 
> 
> > > Problem settled! I love you my dear sister. Since it is still confidential, I will just keep it for personal use.
> > 
> > 
> > What is confidential?
> 
> The file of course. The file above mentioned hasn't been released officially, so it is confidential and I will keep it just for personal use.

myth-busted as anyone that has a proper compiling environment can get that file

TuxSH, 2020-05-22:
This build ""solving"" the issue can only mean you have stuck buttons, which needs repair.

closed 2020-05-22 by TuxSH

JudyLucy, 2020-05-23:
Ok. I see. I will check it when it is convient to me. Thanks for telling me the hardware problem.