LumaTeam/Luma3DS #1427: Very specific Luma3DS crashes when using Checkpoint cheats on some quests of Monster Hunter Generations

Ina-Exter, 2020-05-20:
**System model:** New 3DS

[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):** SYS 11.13.0-45E

[e.g. 11.13.0-45U SysNAND, 11.13.0-45E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):** Boot9Strap

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:** v10.1.3 stable

[e.g. v10.1.3 stable or if using non-releases specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/0543c208fd154e6326ea5da8cbf66ffcbdef010c]

**Luma3DS configuration/options:**

Default EmuNAND: (N/A)

Screen brightness: ( 4 )

Splash: ( Off )

Splash duration: ( 1 )

PIN lock: ( Off )

New 3DS CPU: ( Off )

--

Autoboot EmuNAND: ( N/A )

Use EmuNAND FIRM if booting with R: ( N/A )

Enable loading external FIRMs and modules: ( No )

Enable game patching: ( No )

Show NAND or user string in System Settings: ( Yes )

Show GBA boot screen in patched AGB_FIRM: ( No )

Patch Arm9 access: ( Option unavailable/not found? )

Set developer UNITINFO: ( No )

Disable Arm11 exception handlers: ( No )

--

**Explanation of the issue:**

In Monster Hunter Generations, players are made to navigate a map made of separated areas to hunt a monster. Changing area requires a loading screen. 
It is possible (as you know) to enable cheats at nearly any point in a game. For this game, I only use 3 cheats, all of which require a key combo.
-Player attack x10 (activate Y+up)
-Show monster HP (activate L+up)
-Paralyze monster (activate L+B)
Selecting those cheats in Luma3DS menu is almost never a problem.

In the game, some quests require you to specifically "capture" a monster (i.e. use some items on it when it is low in HP), while you may ordinarily kill it. 
During those quests and with active cheats, **going through any area loading screen will trigger a Luma3DS exception and core dump.**
Additional detail: This has only be tried with Capture quests for "Deviant monsters", a specific category of monsters in the game.

**Steps to reproduce:**

1. Load up any key-combo cheat (not tried with always active cheats) in Checkpoint and start MH Generations

2. Select a Special Permit Capture quest (usually quest number II or VII, note that completion of some previous quests may be required on a fresh save)

3. Once the quest is started, activate Luma3DS menu and select one or more cheats

4. Change areas and trigger the core dump

**Dump file:**

3 dump files available, not all on the same quest.

[mhg_luma3ds_dumps.zip](https://github.com/LumaTeam/Luma3DS/files/4657204/mhg_luma3ds_dumps.zip)

MisterPrintf, 2020-06-17:
This is most likely an issue with the cheat codes rather than Luma3DS's engine.
If cheat codes attempt to edit protected areas in memory, Luma will crash.
To circumvent this, use NTR and CTRPF, which allows writing to all sectors.

suppai, 2022-06-05:
Closing as a tech support question unrelated to Luma3DS itself; the issue tracker is not for tech support, try Nintendo Homebrew's assistance channels (https://discord.gg/C29hYvh).

closed 2022-06-05 by suppai