LumaTeam/Luma3DS #1429: [Request] Per-Game New 3DS Settings

CatsLover2006, 2020-05-24:
I’m not 100% sure what I’m looking for isn’t already in some way implemented, but couldn’t it be possible to override the increased clock and l2 for certain games?
Maybe include a list of prebuilt settings for games which glitch under these circumstances?
**Note:** I’m not including New 3DS enhanced titles, but if it makes life easier, it’s ok if those are added.

CatsLover2006, 2020-05-24:
**Edit:** this could also have to do with the Rosalina menu, like saving the wanted clock and L2 settings to the game.

nayief94, 2020-05-25:
hi

TuxSH, 2020-05-25:
> Maybe include a list of prebuilt settings for games which glitch under these circumstances?

Just disable the corresponding main Luma menu option for it and use the "dynamic" Rosalina option instead.

closed 2020-05-25 by TuxSH

RinMaru, 2020-05-29:
> > Maybe include a list of prebuilt settings for games which glitch under these circumstances?
> 
> Just disable the corresponding main Luma menu option for it and use the "dynamic" Rosalina option instead.

Whats the Dynamic option?