LumaTeam/Luma3DS #1432: Splash Screen Fragments Cause Screen Flickering if Game Patching is Enabled

Meckload, 2020-06-03:
**System model:**

[New 3DS XL]

**SysNAND version (+emu/redNAND version if applicable):**

[11.13.0-45E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[Boot9Strap]

**Luma3DS version:**

[v10.1.3 stable]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (4)

Splash: (Before)

Splash duration: (3)

PIN lock: (Off)

New 3DS CPU: (Clock+L2)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: (x)

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch Arm9 access: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

--

**Explanation of the issue:**

Enabling the Game Patching and any Splash Screen option leads to an inconvenient flickering of the screen until an app is started and exited. 

**Steps to reproduce:**

1. Enable Game Patching and any splash screen option besides "off".

(1.5 the game patching is used for the enhanced Majora's Mask patch)

2. During the booting process, lower fragments of the splash screen are displayed

3. On the home screen, the lower fragments of the splash screen and the home screen are displayed simultaneously, causing the screen to flicker.

4. The flickering of the screen stops after opening and closing an app

5. If the splash screen is disabled, fragments of the Luma settings are shown. Also leading to flickering. Another reboot is necessary to counteract the issue.

**Dump file:**

Halvorsen256, 2020-06-04:
Are you able to reproduce this on video while filming at 60fps? The flickering I know about can't be seen at lower framerates, nor can it be screenshotted.

Meckload, 2020-06-04:
Here is the video at lower fps. The fragments are still visible, but (as you suggested) it is not possible to see the flickering. I was able to film at a higher fps and make the flickering visible with the slow-mo function of my phone's camera. But I'm still trying to figure out how to upload this on a platform (imgur apparently isn't the biggest fan of the slow-mo format).

https://imgur.com/a/CMa5QVX

Halvorsen256, 2020-06-04:
You can try a site like YouTube, right? It'll support 60fps as well.

Meckload, 2020-06-04:
https://youtu.be/xVypCm3jO1Y

There you go. I don't think the video now has 60fps. However, I enabled the slow-mo around 00:50 mark. It makes the flickering very obvious.

Let me know if I should still try to figure out a way to upload it in regular 60fps or if there is anything else that would be helpful.

profi200, 2020-06-05:
This is a screen init/deinit problem. It has been solved but it's not in Luma yet.

Meckload, 2020-06-06:
Okay! Thanks for the update. Amazing to hear it's solved already. Should I keep the issue open until it's officially in Luma or close it?

profi200, 2020-06-06:
Leave it open. The Luma3DS devs will look into it when they have time.

AuroraWright, 2020-11-17:
Main issue: https://github.com/LumaTeam/Luma3DS/issues/1396
Closing

closed 2020-11-17 by AuroraWright

StefanoLassandro, 2022-07-22:
I'm still experiencing this issue on Luma v11.0.
Has the fix been implemented into a release yet?