LumaTeam/Luma3DS #1433: Arm11(core 1) Crash Error a few seconds after boot

HeroMines, 2020-06-06:
**System model: Old 3DS**

**SysNAND version (+emu/redNAND version if applicable): 11.3.0-36U SysNAND**

**Entrypoint (How/what you're using to boot Luma3DS): B9S**

**Luma3DS version: v10.1.3 Stable**

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( 4)

Splash: (After )

Splash duration: (1)

PIN lock: (8)

--

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ()

Patch Arm9 access: ()

Set developer UNITINFO: ()

Disable Arm11 exception handlers: ()

--

**Explanation of the issue:**
I can literally do nothing, and my 3DS will crash for seemingly no reason? 

**Steps to reproduce:**

1. Literally do nothing

2. Profit!

**Dump file:**

[crash_dump_00000018.zip](https://github.com/LumaTeam/Luma3DS/files/4741003/crash_dump_00000018.zip)

TuxSH, 2020-06-07:
From what I know, this bug is cause by Nintendo themselves rather than Luma

Can you try the following:

- Enable "Disable Arm11 exception handlers"
- Upgrade to the latest system version (11.13)
- Disable "Disable Arm11 exception handlers"

HeroMines, 2020-06-10:
Alright, I'll give it a try.

closed 2020-06-18 by HeroMines