LumaTeam/Luma3DS #1434: Pokemon Ultra Moon crashes after going to Poke Pelago

stressedInsomnia, 2020-06-07:
**System model:**

New 2DS XL

**SysNAND version (+emu/redNAND version if applicable):**

11.13.0-45U

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

v10.1.3 stable

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (4)

Splash: (Off)

Splash duration: (1)

PIN lock: (Off)

New 3DS CPU: (Off)

--

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

--

**Explanation of the issue:**

In Pokemon Ultra Moon, the moment the animation of getting on Charizard to go to Poke Pelago finishes, the game crashes.

**Steps to reproduce:**

1. Play Pokemon Ultra Moon

2. Go to Poke Pelago through the menu.

3. After the animation finishes, the game crashes.

**Dump file:**

[crash_dump_00000007.zip](https://github.com/LumaTeam/Luma3DS/files/4741420/crash_dump_00000007.zip)

Shadow322, 2020-06-07:
CheckPoint dump and provide your save

stressedInsomnia, 2020-06-07:
Save:
[main.zip](https://github.com/LumaTeam/Luma3DS/files/4741993/main.zip)

Laximas, 2020-07-19:
Same issue but with different setup. Tried 2 different ROMs, crashes on both.

**System model:**

Old 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.13.0-45E

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

v10.2 stable

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (4)

Splash: (Off)

Splash duration: (1)

PIN lock: (Off)

New 3DS CPU: ( )

--

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

--

Dump file:

[crash_dump_00000004.zip](https://github.com/LumaTeam/Luma3DS/files/4943782/crash_dump_00000004.zip)

Checkpoint save file:

[20200719-155702.zip](https://github.com/LumaTeam/Luma3DS/files/4943784/20200719-155702.zip)

AuroraWright, 2020-11-17:
Confirmed happening on stock, Game Freak's problem

closed 2020-11-17 by AuroraWright

yamijoka, 2023-05-26:
Is there easy way to fix this issue I'm also having this issue