LumaTeam/Luma3DS #1435: Playtime of DS mode games gets not recorded on Luma 10.1.3

EnterpriseFreak-v2, 2020-06-11:
**System model:**
New 3DS "classic" (KTR-001)

**SysNAND version (+emu/redNAND version if applicable):**
11.13.0-45E

**Entrypoint (How/what you're using to boot Luma3DS):**
Luma installed in FIRM0 and FIRM1

**Luma3DS version:**
10.1.3 (Bug occurs in this version)
9.1 (Works in this version)

**Luma3DS configuration/options:**
Screen brightness: 4
Splash: Off
Splash duration: 1
PIN lock: Off
New 3DS CPU: Off

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )
Show NAND or user string in System Settings: ( )
Show GBA boot screen in patched AGB_FIRM: ( )
Patch Arm9 access: ( )
Set developer UNITINFO: ( )
Disable Arm11 exception handlers: ( )

--

**Explanation of the issue:**
When playing DS games on a console with Luma 10.1.3 (and possibly other 10.X releases) play time of these games gets not recorded. Only play time is affected by this, the number of game launches still get recorded properly. This issue occurs not with Luma 9.1 or in OFW.

**Steps to reproduce:**

1. Start any DS game 

2. Let the game run for a minute or two.

3. Return to the home menu and launch the Activity Log App, check the total playtime of the played game which either will not have increased (games that already had playtime recorded) or will be 0:00 (games played for the first time with a version affected by this bug).

Ammako, 2020-06-15:
Works fine here on O3DS and booting from SD. I don't have N3DS, so other N3DS owners would need to check and see if it's a N3DS-specific problem.

For the time being, I think it would be a good idea to switch to a more standard boot method (b9s/fb3ds in firm0) and see if it still happens when you're booting from SD. You can boot from firm0 if you'd like to, but this would help figure out if booting luma from firm0 is what's causing the problem, so others can reproduce it and have an easier fixing it.

If that works, and you absolutely want to boot from firm partition instead of SD, you might be able to get around the problem by flashing fastboot3ds to firm0, luma to firm1, and telling fb3ds to autoboot from firm1. When you do that, Luma doesn't recognize it's being run from firm1 and it just thinks you're running it from SD.

EnterpriseFreak-v2, 2020-06-15:
Just installed B9S and checked if the playtime gets properly recorded now. It does. It looks like this only happens when Luma is installed as FIRM0.

Johnny56732, 2020-06-20:
Certain DS games do that for you I'm playing luminous Arc and it keeps track of how long I'm playing when I save the game

AuroraWright, 2020-11-17:
So is this only an issue when you install as FIRM0? We don't really support that (it's really puzzling though as it shouldn't affect anything)

AuroraWright, 2020-11-17:
Can't reproduce even from FIRM0

closed 2020-11-17 by AuroraWright