LumaTeam/Luma3DS #1439: [Feature Request] TitleID Spoofing

MisterPrintf, 2020-06-17:
I've thought about this for a while and see no reason why it can't be implemented.
What if we could spoof title information?
For example, send Mario Kart 7's TitleID to Nintendo instead of the Homebrew Launcher's?
While I don't know whether this is possible, the implementation user-wise could be rather simple.
For example, a file labelled 'spoof.cfg' inside the luma folder, with lines like this:

[TitleID of application]:[Spoofed TitleID]

This would also allow the 'show friends what I'm playing' setting to be left on, as it would simply show a legit game instead of a homebrew app, provided the user had configured Luma to do so.

Dimensional, 2020-06-17:
Interesting idea. It alone wouldn't be used to get around bans, just make it harder for Big N to find invalid Title IDs and subsequently ban people for those. It wouldn't do anything for people who already were banned, such as myself, so they'd have to use other methods that Luma doesn't want to implement, however this method could be used to at least allow homebrew to run from system installation without the risk of banning.

Oreo639, 2020-06-17:
It is possible to use offical applications as the Homebrew Launcher.
There is an option in Rosilina to switch the current title to HB. Then go to `save settings`.
Your change will persist across reboots.

Dimensional, 2020-06-17:
That's for the homebrew menu. What about running homebrew from the home screen after they were installed from cia files, like some save editors that require extra access not possible in the homebrew menu, or emulators like RetroArch PSX which only run well if loaded from the home screen? Or even some VC injections?

Oreo639, 2020-06-17:
Installing homebrew as CIAs in order to get custom permissions isn't necessary with rosilina.

But you are right I wasn't thinking about that.

TuxSH, 2020-07-05:
This would require NS to be reimplemented, in other words, this is a PITA :)

closed 2020-07-05 by TuxSH