LumaTeam/Luma3DS #1440: [Rosalna[Feature Request]

Rajveer86, 2020-06-20:
**System model:**

[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.13.0-45U SysNAND, 11.13.0-45E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. v10.1.3 stable or if using non-releases specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/0543c208fd154e6326ea5da8cbf66ffcbdef010c]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch Arm9 access: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

--

**Explanation of the issue:**

**Steps to reproduce:**

1.

2.

**Dump file:**

TuxSH, 2020-06-30:
it's kinda hard because you have to deal with APT for app deinit->home menu->app, that's why I've not implemented it.

You could use hbmenu's netloader (press X) with `3dslink`

Rajveer86, 2020-07-01:
Yeah that's fair enough. I was trying to simplify the development process and find a flow where I could just build, push to the 3ds and start debugging all in one step to make the development process easy. I'm almost there but the problem I'm facing is that every time I have to re-enable "Force-debug next application at launch" before pressing Y, and then I can build, push and attach the debugger. It doesn't sound like much, but doing it every time is a bit of a pain point :)

Maybe another way to approach this is to adjust how the "Force-debug next application at launch" works, so that it's always/automatically enabled under certain conditions (e.g. auto on for a specific process name, 3dsx filename e.t.c)?

SeleDreams, 2023-02-05:
I'm having the same issue, i don't know if there has been any progress since then, but this workflow of restarting the homebrew launcher > go to rosaling and enable debug next app at launch all the time is very painful for a development workflow which is why i end up using citra for development at the moment

it would be nice to have something similar to psplink on the psp