LumaTeam/Luma3DS #1442: Adding more option to screen filter.

xpirad, 2020-06-22:
The screen filter function is very useful on my 2dsxl.
However, my 3dsxl screen is kind of yellow tinted. even without battery saver turned on..
I've searched the internet and there are many cases just like mine, and there seems to be no solution about it up to date.
It's like, blue light filter is turned on by default even i'm not enabling it.
Will it be able to add more options for higher color temperature?
Like ,the default is 6500k, can it be raised to 7500k?
Thanks a lot!!

profi200, 2020-06-22:
There are 2 problems with this request. Let me explain:

1. Imagine the 3DS displays a picture with the maximum blue component brightness which is 255 (8 bit). You want more blue but blue is already maxed out. This leads to problem 2.
2. If we lower green and red instead we get the effect we want but at the cost of color accuracy/range.

These panels are probably badly calibrated from the factory but it's unknown how to correct this on the panel side. Doing the correction on the 3DS side has these 2 problems mentioned above.

xpirad, 2020-06-23:
Thanks profi200! But can you check out the page here?
https://gbatemp.net/threads/ctr_redshift-hardware-based-blue-light-filter-for-old3ds-and-2ds.493736/
I've downloaded the cia and tuned up color temperature and it works.
However, the thing is that this setting only works in that app and once i quitted it, everything's just like before..
I really wish this problem can be solved, it's kind of annoying...
Thank you again!!

xpirad, 2020-06-23:
Also, there is this post on page 7.
https://gbatemp.net/threads/ctr_redshift-hardware-based-blue-light-filter-for-old3ds-and-2ds.493736/page-7
A guy named 8bitwonder posted something like this

[image]

Maybe this can be merged into luma? (with more options like 7000k, 7500k ,etc..)

profi200, 2020-06-23:
Yeah, i have not looked too closely how it's implemented. I seems like it's working but you will certainly run into 1 of the mentioned problems. 2 is probably the lesser evil.

xpirad, 2020-06-23:
I'd still prefer that instead of getting a weird yellow tint :(
Thanks for the help anyway!

profi200, 2020-06-23:
It's not up to me anyway to decide if this is accepted. Just wanted to give some input.

(And i would not mind a slight yellow tint. It makes the picture warmer and therefore more eye-friendly. My N3DS XL actually has that on the IPS top screen.)

xpirad, 2020-06-23:
Yeah, but i just want to say thank you anyway.
I know that warmer color is better to the eyes, but i'd still want to at least have a choice on that. It can be annoying sometimes if something that should be white is yellow-ish :/

TuxSH, 2020-06-30:
Is this just a matter of changing the Color Temperature, or does the algo needs to be changed?

xpirad, 2020-07-01:
@TuxSH  i suppose just the color temperature will do.

commit e69f89a0 (2020-07-05): Change screen filters presets (adding more) menu & print CCT. Fixes #1442.

Most new presets come from 8bitwonder
closed 2020-07-05 by TuxSH