LumaTeam/Luma3DS #1445: Add option to keep wifi enabled during GBA games and other virtual console games

electroniclogic, 2020-06-29:
I am trying to capture the screen while playing a gba game and the wifi shuts off, disabling the wireless capture. If you could somehow get this to keep wifi on it would be of great help.

RocketRobz, 2020-06-29:
I believe it would also require Rosalina to keep running in the background, which afaik, isn't possible.
Same applies to DS(i) mode.

Shadow322, 2020-06-29:
not gonna happen for GBA as when its running in GBA mode it is on bear hardware as GBA never had wifi

profi200, 2020-06-29:
Not doable with official AGB_FIRM. Neither can Rosalina run under AGB_FIRM nor is WiFi fast enough to transfer RAW 240x160 frames. Not to mention there is barely enough memory available because FCRAM is disabled.

Capturing to the SD card is **maybe** possible but definitely not with official AGB_FIRM.

TuxSH, 2020-06-30:
As profi said.

You're better off using something like mGBA on your PC for that use case instead.

closed 2020-06-30 by TuxSH