LumaTeam/Luma3DS #1446: Inconsistent crash that happens when pressing home whilst in the middle of a game

Despresso99, 2020-06-29:
[image]

Hello, I seem to encounter this bug and I am not sure what is the cause of it. Essentially, sometimes when I press home whilst in the middle of a game, the 3ds crashes and leads me to that error message. It does not happen all the time, but only sometimes. Attached is the crash dump.

[crash_dump_00000003.zip](https://github.com/LumaTeam/Luma3DS/files/4844118/crash_dump_00000003.zip)

Despresso99, 2020-06-30:
If any additional information is needed to be provided, I can happily provide. :)

Despresso99, 2020-07-12:
I ended up fixing it! 
The issue was that I was using a ROM Hack for my Pokemon game that only supported the base version of the game - the updates interfered with it working. Removing the game update seems to have fixed it. :)

suppai, 2022-06-05:
Closing as a tech support question unrelated to Luma3DS itself; the issue tracker is not for tech support, try Nintendo Homebrew's assistance channels (https://discord.gg/C29hYvh).

closed 2022-06-05 by suppai