LumaTeam/Luma3DS #1449: Luma3ds payload dont work

scepzinho, 2020-07-10:
Luma3ds payloads is not working and so GM9 is not building when i press start while opening it and dont ask if my folder is named right and stuff like that because i checked all it :P

TuxSH, 2020-07-11:
Do you have more than one payload?

scepzinho, 2020-07-12:
No

TuxSH, 2020-07-12:
If you only have one payload it won't show up the payload menu and will instead launch the payload directly

closed 2020-07-12 by TuxSH

scepzinho, 2020-07-12:
But it show nothing doesnt show  payload and doesnt show gm9 (boot in
payload folder)

scepzinho, 2020-07-12:
Bro you shouldn't close it, my problem is not solved :/

TuxSH, 2020-07-13:
This is a GM9 bug, https://github.com/d0k3/GodMode9/issues/578, not my issue.

Try to have more than one payload in the payload folder, I guess.