LumaTeam/Luma3DS #1452: compiling error

Shadow322, 2020-07-11:
when I type "make release" I get this error and only has happen since the past 3-4 times new code got pushed

"make: *** No rule to make target 'boot.3dsx', needed by 'Luma3DSv10.1.3-e3bb1c1b.zip'.  Stop."

DMSalesman, 2020-07-12:
Currently running dc67d438dc43e3b8192b035cad2b742eb802cfc2 which built fine yesterday.

Shadow322, 2020-07-12:
> when I type "make release" I get this error and only has happen since the past 3-4 times new code got pushed
> 
> "make: *** No rule to make target 'boot.3dsx', needed by 'Luma3DSv10.1.3-e3bb1c1b.zip'. Stop."

Since your ignorant of the error butt out

DMSalesman, 2020-07-12:
First - I noticed my oversight and removed the message.

Second - consider hiring a PR manager, if you're unable to follow some basic etiquette.

Thank you for your time!

Shadow322, 2020-07-12:
> First - I noticed my oversight and removed the message.
> 
> Second - consider hiring a PR manager, if you're unable to follow some basic etiquette.
> 
> Thank you for your time!

First - consider hiring a English teacher so u can read
Second - butt out

closed 2020-07-12 by Shadow322