LumaTeam/Luma3DS #1456: No sleep with Homebrew Launcher in the background

Nemris, 2020-07-14:
**System model:**

[e.g. 2DS, New 3DS, Old 3DS]
New 3DS
**SysNAND version (+emu/redNAND version if applicable):**
11.13.0-45E
[e.g. 11.13.0-45U SysNAND, 11.13.0-45E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap
[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**
https://github.com/LumaTeam/Luma3DS/commit/362c4ffff13509d1aca3f40b1e221ff397233af3
v10.1.3 stable
v9.1 stable
[e.g. v10.1.3 stable or if using non-releases specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/0543c208fd154e6326ea5da8cbf66ffcbdef010c]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (1)

Splash: (before)

Splash duration: (1)

PIN lock: ( )

New 3DS CPU: (Clock+L2)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: (x)

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch Arm9 access: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

--

**Explanation of the issue:**
When the Homebrew Launcher is paused in the background, the system won't be able to enter a proper sleep mode, as evidenced by the Power LED no blinking and the battery consumption staying high.

The issue happens both in the Home menu and in the top-row 3DS applets such as Friend List.

I wasn't able to test Luma versions older than v9.1 due to the 3DS' OS being too recent. If this is an actual Luma bug, it might be older than that.

**Steps to reproduce:**

1. Access the Homebrew Launcher, either with a frontend .cia or via Rosalina's `Switch the hb. title to the current app`;

2. press the Home button to pause the Launcher;

3. enter sleep mode and look at the Power LED.

**Dump file:**

TuxSH, 2020-07-14:
This is a libctru bug that was fixed but not released. Next release of stuff (which should come soon) will fix it.

fincs, 2020-07-14:
This is not a Luma3DS bug. It is a libctru bug that has already been fixed (see https://github.com/devkitPro/libctru/commit/7171a87d53fe3998e76d380707f3a6cac54c074c). The next release of libctru and 3ds hbmenu will contain this fix.

Nemris, 2020-07-14:
I'll try building Luma against the libctru HEAD, in that case.

fincs, 2020-07-14:
It's not related to Luma at all. It's a bug in libctru's APT code, that is present on *every single* homebrew application. Building Luma from source won't help at all.

Nemris, 2020-07-14:
I did understand that the bug is in one of the dependencies itself, yes - that's why I plan to build it myself, and then have my own "fixed" libctru and Luma while I wait for a release. It's more of an excuse to tinker rather than anything else, really.

Thank you for clarifying the matter, though! I'll close the issue, now.

closed 2020-07-14 by Nemris

fincs, 2020-07-14:
Luma doesn't even make use of libctru's apt code, so recompiling it will accomplish nothing...

Nemris, 2020-07-14:
So, if I understood this, what I'd need to recompile was libctru itself, the new-hbmenu `boot.3dsx` and _every_ other .3dsx I use. Is that right?

fincs, 2020-07-14:
Yes. The next release will happen soon, so it would be best appreciated if you let things run its natural course instead of recompiling random things, though.