LumaTeam/Luma3DS #1457: Chainloader no longer works with 10.2

bmn001, 2020-07-16:
**System model:**
New 2DS XL

**SysNAND version (+emu/redNAND version if applicable):**
Sys 11.13.0-45U

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap

**Luma3DS version:**
v10.2 stable

**Luma3DS configuration/options:**

Screen brightness: (4)

Splash: (off)

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

**Explanation of the issue:**

I upgraded from Luma 10.1.3 to 10.2.

All went well.

However, I'm now unable to boot to the chainload menu.

When I power-up holding the Start button, I get a black screen that doesn't respond to inputs.

I only had one file inside of /luma/payloads, and that's GodMode9.firm (the latest, v 1.9.1)

I thought "Okay, maybe that means GodMode9 isn't compatible with Luma 10.2 yet".  So I deleted it, leaving /luma/payloads empty.

However even with an empty payload dir, holding Start when powering on will still result in a black screen that I need to power-off to resolve.

**Steps to reproduce:**

1. Fully power off on Luma 10.2.

2. Power on holding Start.

**Dump file:**
n/a

MojabeJoe, 2020-07-16:
Same here. On Old 3DSXL - 11.13.0-45U.

aspargas2, 2020-07-16:
I can't reproduce the part where it happens with an empty payloads directory, and GodMode9 not booting is a GodMode9 issue.

Oreo639, 2020-07-16:
Chainloader still works in 10.2. Just not properly with godmode9 1.9.0 or later.

You can workaround this by making a copy of godmode9.firm in the payloads folder but with a different name.

> However even with an empty payload dir, holding Start when powering on will still result in a black screen that I need to power-off to resolve.

I can't reproduce this on an O3DS. With an empty payloads dir, it gives me a black screen untill I release the `start` button where it boots into the OS as normal.

MojabeJoe, 2020-07-16:
> 
> 
> Chainloader still works in 10.2. When using it with gm9 1.9.0 or later without going to the payload selection screen, the backlight isn't initialized correctly.
> 
> You can workaround this by making a copy of godmode9.firm in the payloads folder but with a different name.
> 
> > However even with an empty payload dir, holding Start when powering on will still result in a black screen that I need to power-off to resolve.
> 
> I can't reproduce this on an O3DS. With an empty payloads dir, it gives me a black screen untill I release the `start` button where it boots into the OS as normal.

THIS WORKED FOR ME. Holding Start on boot immediately brought up the two files. Thanks a billion.

closed 2020-07-16 by TuxSH

bmn001, 2020-07-17:
I see this was closed with no comment.  Is this...not a bug?  Am I doing something wrong?  This worked with 10.1.3.

Shadow322, 2020-07-17:
> I see this was closed with no comment. Is this...not a bug? Am I doing something wrong? This worked with 10.1.3.

Because we can't reproduce it as it works here

profi200, 2020-07-17:
It's related to #1396 but because the bug happens with the screens never turned on until GodMode9 is reached it's on GodMode9's side. But this bug really affects all bare metal homebrews doing screen init.

T0biasCZe, 2020-08-17:
i have that issue too, New 3DS Xl, and the screen is black

T0biasCZe, 2020-08-17:
I found a workaround: instead of godmode, start luma menu (select), and then press START 3times QUICKLY and keep it pressed on the 3rd press
it will start
its backlight start issue

Shadow322, 2020-08-17:
> I found a workaround: instead of godmode, start luma menu (select), and then press START 3times QUICKLY and keep it pressed on the 3rd press
> it will start
> its backlight start issue

It has already been fixed in latest source

HY-SANG, 2020-08-27:
> I found a workaround: instead of godmode, start luma menu (select), and then press START 3times QUICKLY and keep it pressed on the 3rd press
> it will start
> its backlight start issue

This works! THANKS!!!

Shadow322, 2020-08-27:
> > I found a workaround: instead of godmode, start luma menu (select), and then press START 3times QUICKLY and keep it pressed on the 3rd press
> > it will start
> > its backlight start issue
> 
> This works! THANKS!!!

It has already been fixed in latest source as I keep saying

T0biasCZe, 2020-08-27:
But you need to manualy compile it, and there can be another bugs

Shadow322, 2020-08-27:
> But you need to manualy compile it, and there can be another bugs
> […](#)

Then it's a good idea to do that